Simba-Avionic/srp #14: Create I2C driver

Mateusz-Krajewski, 2024-01-02:
TODO
- [ ] Co to I2C
- [ ] Jak działa I2C
- [ ] Jak korzystać z I2C na Rpi4 lub beaglebone bez korzystania zewnętrznych bibliotek
- [ ] stworzenie interfejsu dla biblioteki I2C
- [ ] zaimplementowanie biblioteki I2C

BACKLOG
- [ ] Wymyślić emulowanie czujników po i2c

Przydatne linki:
- https://medium.com/geekculture/raspberry-pi-c-libraries-for-working-with-i2c-spi-and-uart-4677f401b584
- https://stackoverflow.com/questions/22611675/i-am-trying-to-use-i2c-on-the-beagle-bone-black-with-c-but-i-keep-getting-0x00
- https://pl.wikipedia.org/wiki/I%C2%B2C

bartoszsnieg, 2024-01-10:
example https://github.com/shenki/linux-i2c-example/blob/master/i2c_example.c

Mateusz-Krajewski, 2024-01-11:
srp/core/i2c_driver

closed 2024-05-14 by Mateusz-Krajewski