Simba-Avionic/srp #63: Config Generator ignore method client in db section

Mateusz-Krajewski, 2024-04-21:
[log.zip](https://github.com/Simba-Avionic/srp/files/15053052/log.zip)
[image]

Jeśli klient  jest na tym samym komputerze co server, config parser nie dodaje jego adresu do sekcji db pliku app_someip.json.
Problem nie występuje jeśli klient i server są na różnych komputerach.
[image]

bartoszsnieg, 2024-04-22:
Widze że App ID dałeś o jeden znak za długi

bartoszsnieg, 2024-04-22:
Po rzeanalizowaniu problemu uważam ze bła jest w skrypcie od generowania plików konfiguracyjnych. Problem rozwiązany będzie wraz z wprowadzeniem wersji 1.5 (Access list)

bartoszsnieg, 2024-04-22:
Ale cały nie powiem dlt logi to cos zajebistego w naszym przypadku ps. nawet sentinel tego nie ma ;)

Mateusz-Krajewski, 2024-04-22:
> Ale nie powiem dlt logi to cos zajebistego w naszym przypadku ps. nawet sentinel tego nie ma ;)

tak, mega ułatwia prace

closed 2024-04-27 by bartoszsnieg